niru-5/imusensor #1: cannot install imusensor - the library seems not available ..

PatrickBustraen, 2020-07-01:
**pip install imusensor** 
results in  
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting imusensor
  Could not find a version that satisfies the requirement imusensor (from versions: )
No matching distribution found for imusensor

Working with a Raspberry Pi4 platform..

niru-5, 2020-07-01:
Hi Patrick,

I am not sure why it could not find the distribution. Could you check the following 
1) Make sure pip is upgraded
2) Are you able to install other packages like smbus, numpy. 
3) Are you under a proxy network, then check out this [link](https://stackoverflow.com/questions/55021270/could-not-find-a-version-that-satisfies-the-requirement-flask-from-versions) once? 
4) Could you try this command -> ```pip install imusensor==1.0.0```
5) if nothing works try -> ```pip install imusensor==1.0.0 --verbose``` and paste the log, I will see if I can figure out something

I have tried it out on my rpi3 and it is working. I don't think there should be any problem with rpi4. 
Let me know if any of the above work. If not the last resort can be to download the repo and include the path to the repo in your examples or usage.

niru-5, 2020-07-01:
An update on the issue. 
The package was intended for python3. 
by default pip will be for python2 as far as I know. 
For installing it with python3 use the command -> ```pip3 install imusensor==1.0.0```
I guess that should work.
In that sense you will have to run the examples by ```python3 imuExtraction.py``` 
Or you could create a virtual environment on python3.

PatrickBustraen, 2020-07-01:
It now works with pip3
I am still working mostly with pip, pip 2
Thanks for helping out, I will test the lib later on, really curious on the quality of the results, thx a lot

closed 2020-07-01 by PatrickBustraen