niru-5/imusensor #3: Madwick filter computation

Fibo27, 2020-07-13:
Hi - when i apply the madgwick filter, the values seem to be out of sync with the base case values or those that of the K filter. While going through Kris Winer's code, he has commented the below quoted points while computing the Quaternions. Is this something that you have considered?

"Sensors x (y)-axis of the accelerometer/gyro is aligned with the y (x)-axis of the magnetometer;
the magnetometer z-axis (+ down) is misaligned with z-axis (+ up) of accelerometer and gyro!
We have to make some allowance for this orientation mismatch in feeding the output to the quaternion filter.
For the MPU9250+MS5637 Mini breakout the +x accel/gyro is North, then -y accel/gyro is East. So if we want te quaternions properly aligned we need to feed into the Madgwick function Ax, -Ay, -Az, Gx, -Gy, -Gz, My, -Mx, and Mz. But because gravity is by convention positive down, we need to invert the accel data, so we pass -Ax, Ay, Az, Gx, -Gy, -Gz, My, -Mx, and Mz into the Madgwick function to get North along the accel +x-axis, East along the accel -y-axis, and Down along the accel -z-axis.
This orientation choice can be modified to allow any convenient (non-NED) orientation convention. Pass gyro rate as rad/s"

niru-5, 2020-07-21:
Hi, 

Could you explain what do you mean by out of sync. Are the values completely different or are they just different by a sign/180 angle. 
Also keep in mind that madgwick takes time to stabilize as it follows a gradient descent approach. unlike kalman, it takes a few iterations to give steady values and then onwards it works smoothly. 

Yes, I did consider the alignment of axes of the sensors. In my case you can see the transformation matrix in config.py the variable ```transformationMatrix```
However, I have modified the accelerometer and gyroscope to match the axes of magnetometer. 
Hence I have converted ax, ay, az of accelerometer as ay, ax, -az. 
This was done similarly for gyroscope as well. See the image below for reference. 

[image]
Image was taken from MPU9250 document. [link](https://invensense.tdk.com/wp-content/uploads/2015/02/PS-MPU-9250A-01-v1.1.pdf)

Note: I tried using changing the directions of accelerometer because accel data is inverse, but I didn't see much effect on the sensor values. I mean yes, the initial orientation changed, however kalman and madgwick were giving the same results.

Fibo27, 2020-07-22:
Ok thanks - that makes sense then. As per my comment on being out of sync - yes it was a change of phase. Thank you for your good work

closed 2020-07-22 by Fibo27

niru-5, 2020-07-22:
Thanks for the appreciation. I will update the library with sensor values extracted from DMP as well. I will follow up with a blog for better understanding in the coming week. Take a look at that time