niru-5/imusensor #4: Wrong message being displayed

gnieradk, 2020-07-13:
You have done a good job with the library. Thank you!
But I have a problem/question with the code in lines 54-56 of the MPU9250 module.
The code which you have written is:
```
		name = self.__whoAmI()
		if not `(name[0] == 113 | name[0] == 115 ):`
			print ("The name is wrong {0}".format(name))
```
So the returning value from sensor is 0x71 which is value equals 113.
If the code is the `(name[0] == 113) `is True; the` (name[0] == 115)` is False so the full condition `(name[0] == 113 | name[0] == 115 ):` is False. `if not ` gives the True value and as a results the print line is executed and in my opinion the line should not be executed (because the sensor has proper value read by `__whoAmI` function.
So my question is it is some mistake or I don't understand something properly?
Thank You for work, help and explanation.

niru-5, 2020-07-21:
Thanks for the compliment. I could not get to the issues last week as I busy with something else. I will soon try to create blogs for easy understanding and use-cases.

In your statement below
If the code is the ```(name[0] == 113)``` is True; the ```(name[0] == 115)``` is False so the full condition ```(name[0] == 113 | name[0] == 115 ):``` is False
"OR" condition will give true if even one of them is true, so even if one of the conditions ```name[0] == 113``` or ```name[0] ==115``` is true, it doesn't matter what the other statement is, it will just be true. The outside negation will make it false and hence the debug statements won't be printed.
table for OR
|Input 1 | Input 2| Output |
|--------|----------| -----|
| 0 | 0 | 0 |
| 0 | 1 | 1 |
| 1 | 0 | 1 |
| 1| 1 | 1 |

gnieradk, 2020-07-28:
Thanks. I agree with You and of course You are right.
I don't know why - but I had to change the OR operator from `|` to the Python `or`
So the checking name line looks:
`(name[0] == 113 or name[0] == 115 )`
And everything works as with the intention.
Probably this is some Python specific thing with bitwise and logical operator but I don't know how to track the difference.
BTW, thanks. You can close this topic.

niru-5, 2020-07-28:
Sure.

closed 2020-07-28 by niru-5