niru-5/imusensor #7: 'name is wrong [113]'

sawan-flo, 2020-12-23:
hello,
this is not particularly an error but when i initially run the kalman filter code, it 'prints' on the console <The name is wrong [113]> after which it takes about a second and then starts giving the roll,pitch,yaw value. Is it something to be concerned about? Also how to remove this message?

Thanks in advance

niru-5, 2020-12-23:
Hi, 
That was a small check to make sure it gets connected to MPU9250. I kind of hardcoded it based on the few sensors. I don't think it should be a problem. 
I didn't put any verbose argument to mask the statement. I am not sure if I can get to it this week. You can open a pr for it, if you like. 

Thank You
Niru

closed 2021-05-07 by sawan-flo

gitthnx, 2023-08-26:
`if (name[0] != 113 | name[0] != 115):`  instead of  `if not (name[0] == 113 | name[0] == 115):` avoids ID 'mismatch' information, e.g. "The name is wrong [115]"