niru-5/imusensor #8: [Errno 121] i2c adresses messed up

DavidD49, 2021-01-31:
I'm using your code on a raspberry pi zero with python 3.7 and the board MPU-9250. 

the problems : 

- the demonstration code leads to `[Errno 121] Remote I/O error` 
- the register of the sensor is messed up after the demonstration code

I installed some modules 
```
sudo apt-get update
sudo apt-get upgrade
sudo apt-get install i2c-tools
sudo pip install smbus
pip3 install easydict
pip3 install imusensor
```
(easydict was not included in the Pre-requisites)

before the demonstration code I probed the i2c. The sensor react as expected.
```
pi@raspberrypi:~ $ sudo i2cdetect -y 1
0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- 0c -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
```

then I get the demonstration code from [https://pypi.org/project/imusensor/#description](url)
``` python
import os
import sys
import time
import smbus

from imusensor.MPU9250 import MPU9250

address = 0x68
bus = smbus.SMBus(1)
imu = MPU9250.MPU9250(bus, address)
imu.begin()
# imu.caliberateGyro()
# imu.caliberateAccelerometer()
# or load your own caliberation file
#imu.loadCalibDataFromFile("/home/pi/calib_real_bolder.json")

while True:
	imu.readSensor()
	imu.computeOrientation()

	print ("roll: {0} ; pitch : {1} ; yaw : {2}".format(imu.roll, imu.pitch, imu.yaw))
	time.sleep(0.1)
```

run it 
```
pi@raspberrypi:~/Desktop/IMU/02_imusensor $ python3 Check_Instal_imusensor.py 
```
get the [Errno 121]
```
Traceback (most recent call last):
  File "Check_Instal_imusensor.py", line 11, in <module>
    imu.begin()
  File "/home/pi/.local/lib/python3.7/site-packages/imusensor/MPU9250/MPU9250.py", line 48, in begin
    self.__writeAK8963Register(self.cfg.Ak8963CNTL1, self.cfg.Ak8963PowerDown)
  File "/home/pi/.local/lib/python3.7/site-packages/imusensor/MPU9250/MPU9250.py", line 551, in __writeAK8963Register
    self.__writeRegister(self.cfg.I2CSlave0Control, self.cfg.I2CSlave0Enable | 1)
  File "/home/pi/.local/lib/python3.7/site-packages/imusensor/MPU9250/MPU9250.py", line 535, in __writeRegister
    val = self.__readRegisters(subaddress,1)
  File "/home/pi/.local/lib/python3.7/site-packages/imusensor/MPU9250/MPU9250.py", line 543, in __readRegisters
    data = self.Bus.read_i2c_block_data(self.cfg.Address, subaddress, count)
OSError: [Errno 121] Remote I/O error
```

Then checked the i2c
```
pi@raspberrypi:~/Desktop/IMU/02_imusensor $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          03 -- -- -- 07 08 09 -- -- -- 0d 0e 0f 
10: -- -- -- -- 14 15 16 -- 18 -- 1a 1b 1c 1d -- -- 
20: -- -- -- -- -- -- -- -- 28 29 2a 2b 2c 2d -- -- 
30: 30 31 -- 33 34 35 36 -- -- 39 3a 3b 3c 3d 3e 3f 
40: -- 41 42 43 44 45 46 -- -- 49 4a -- -- -- -- 4f 
50: -- -- 52 -- 54 55 56 57 -- -- -- -- -- 5d -- 5f 
60: -- -- -- -- -- -- -- -- 68 -- 6a -- -- -- 6e 6f 
70: 70 71 72 -- 74 75 76 77                         
```
could not believe it so rechecked :
```
pi@raspberrypi:~/Desktop/IMU/02_imusensor $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          03 04 -- 06 -- -- -- 0a -- 0c -- 0e -- 
10: -- -- 12 13 14 15 -- 17 -- 19 -- 1b -- -- 1e 1f 
20: 20 -- -- -- -- 25 -- -- -- 29 2a 2b 2c -- 2e -- 
30: -- 31 32 -- 34 -- -- 37 -- 39 -- -- -- -- -- -- 
40: -- -- -- -- -- -- 46 -- -- -- -- -- 4c -- 4e 4f 
50: -- -- -- 53 54 55 -- 57 58 -- -- -- -- -- -- -- 
60: -- 61 -- -- -- 65 66 67 68 69 6a 6b 6c 6d 6e 6f 
70: -- 71 72 73 -- 75 -- --     
```                    

It seems that the sensor register is not responding as expected :
- addresses changes from call to call 
- to much addresses

Because there are not many questions about [Errno 121]. I think I'm missing something, there is a high probability the error is on my side.

So my questions are: 

- What am I missing?
- What do I have to do to get my sensor register back?

DavidD49, 2021-02-03:
Solved, 
I wanted to test different methods to access the MPU 9250 and later select the one more adapted to my project. Among those methods there was an increased baudrate with the i2c. I thought I uninstalled it. but :
i2c_arm_baudrate=1000000 
so correction :
```
sudo nano /boot/config.txt
dtparam=i2c_arm=on,i2c_arm_baudrate=400000
```
then reboot
```
pi@raspberrypi:~ $ sudo i2cdetect -y 1
0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- 0c -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- 
```
solved

closed 2021-02-03 by DavidD49

rom1es, 2021-04-04:
Hi DavidD49, 
I have exactly the same issue than you. 
Setting i2c_arm_baudrate at 400000 or 1000000 don't change the result...  
Do you modified something else ?
Thanks for your help.

DavidD49, 2021-04-04:
Hi rom1es, 

I didn't continue with this library, I wrote a simpler code inspired by this project : 
[https://github.com/kriswiner/MPU9250](url)

rom1es, 2021-04-05:
Hi. thanks for this link, the chriswiner's package is full of informations.
I hope I have solved my problem. 
I've detected that the WHO_AM_I for AK8963 was 0xFF instead of 0x48

$ i2cget -y 1 0x0C 0x00
0xff
$ 

Maybe the my AK8963 device is configured as I2C slave. I don't know why because a few days ago, I saw 0x48 at this WHO_AM_I address. 

The imusensor library access to the AK8963 device as slave I2C device.
So, I've unplug EDA and ECL wires on my breadboard and restart the test program with imusensor, and "voilà", I have data !

$ python3 imusensor_basic.py 
The name is wrong [113]
roll: -177.0002441131415 ; pitch : -1.3819353125379346 ; yaw : 8.478782566942568
roll: -176.79302391074728 ; pitch : -1.2005549371737165 ; yaw : 7.9572838846634175
roll: -176.99525694845804 ; pitch : -1.5569402761370297 ; yaw : 7.7464107377321945
roll: -176.50979785732198 ; pitch : -1.2092709494596665 ; yaw : 8.075844930367289
roll: -177.000344693504 ; pitch : -1.2833273776345917 ; yaw : 7.046193712212783

And without seen anymore the device 0x0C on the I2C bus:

$ i2cdetect -y 1 
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
$ 

Hope it will be useful for someone else.

niru-5, 2021-04-11:
> Hi rom1es,
> 
> I didn't continue with this library, I wrote a simpler code inspired by this project :
> [https://github.com/kriswiner/MPU9250](url)

This library was inspired by Kriswiner's work as well(mentioned in acknowledgements). His repo is amazing to understand some of the concepts related to IMU.

masterbuchi, 2023-08-08:
Having the same problem, another script runs totally fine with the IMU, but with this one its creating weird configurations in I2C and I am getting Remote I/O errors.

Before:
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- 0c -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- -- 

After:
    0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- 0c -- -- 0f 
10: -- -- -- -- -- 15 16 -- -- 19 -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- 46 -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- 54 -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         

@DavidD49 Do you still have your code? Cause I am close to throwing this thing out the window...