niru-5/imusensor #12: Connecting i2c to EDA/ECL

bernardyip, 2021-04-11:
Hi niru

Thank you for working on this library! It has made working on my project much easier.

Just wanted to double check if there is a need to connect ECA/ECL lines on the MPU9250 to the SCL/SDA of my RPI as that contains information of the magnetometer? I've seen some other tutorials that requires us to do this but in the README it doesn't say that.

Found this from here: https://makersportal.com/blog/2019/11/11/raspberry-pi-python-accelerometer-gyroscope-magnetometer

niru-5, 2021-04-11:
Hi, 

As far as I understand, EDA/ECL lines are for including mag in the I2C communication as well. However, the library does this through the MPU, so I don't see the need to include EDA/ECL lines. 

Thank You
Niru

bernardyip, 2021-04-11:
Sounds good! Thanks!

closed 2021-04-11 by bernardyip