niru-5/imusensor #14: Can´t read the magnetometer

JGonzalga, 2021-06-21:
Hello, I am trying this code for the  MPU9250 and I am able to obtain the pitch and roll orientations, but I am not able to read the magnetometer for the yaw estimation. This is what I get when running the script:
[image]

It seems that the script is not able to configure the magnetometer, when looking into its registers this is what I found:

[image]

Neither the Id and the name of the sensor seems right, do you have any advice on how to fix it? 

Thanks in advance 

Josué

niru-5, 2021-06-21:
Hi, 

Did you calibrate your Mag sensor? If yes, did you rotate the sensor in 8 direction while calibration? Were you getting any values without calibration? 

The issue seems similar to [this](https://github.com/niru-5/imusensor/issues/13)

JGonzalga, 2021-06-21:
Thanks for your response!
I tried it with the imu.caliberateMagApprox()  function but it was not successful, this was what I got: 
[image]

I was not getting any yaw data before the calibration attempt.

roboteer-HJ, 2021-07-06:
@JGonzalga Try using mag precise  calibration and while calibrating move your sensor in **∞** (infinity) shape.
I also faced the same problem and was able to rectify using the above method.

marcofariasmx, 2022-05-15:
Were you able to fix this issue? I'm still having the same trouble and when the mag calibrates I get the following error 
[image]

niru-5, 2022-05-15:
Hi @marcofariasmx,

In MagApprox()
Looks like the values captured by mag during the calibration process do not change much. 
The code for the function is [here](https://github.com/niru-5/imusensor/blob/488aebe2dec012f1813ab94195afa7e8af177814/imusensor/MPU9250/MPU9250.py#L350)
We get nan values when the min and max values during the calibration process end up zero. 
```
import smbus
import numpy as np
import sys
import os
import time

from imusensor.MPU9250 import MPU9250

address = 0x68
bus = smbus.SMBus(1)
imu = MPU9250.MPU9250(bus, address)
imu.begin()
print ("Accel calibration starting")
imu.caliberateAccelerometer()
print ("Accel calibration Finisehd")
print (imu.AccelBias)
print (imu.Accels)

def caliberateMagApproxDebug(imuObj):
	"""Caliberate Magnetometer
	
	This function uses basic methods like averaging and scaling to find the hard iron
	and soft iron effects.
	Note: Make sure you rotate the sensor in 8 shape and cover all the 
	pitch and roll angles.
	"""

	currentSRD = imuObj.CurrentSRD
	imuObj.setSRD(19)
	numSamples = 1000
	magvals = np.zeros((numSamples,3))
	for sample in range(1,numSamples):
		imuObj.readSensor()
		magvals[sample] = imuObj.MagVals/imuObj.Mags + imuObj.MagBias
		time.sleep(0.02)
	minvals = np.array([magvals[:,0].min(), magvals[:,1].min(), magvals[:,2].min()])
	maxvals = np.array([magvals[:,0].max(), magvals[:,1].max(), magvals[:,2].max()])
    print ("min vals ", minvals)
    print ("min vals ", maxvals)

	imuObj.MagBias = (minvals + maxvals)/2.0
	averageRad = (((maxvals - minvals)/2.0).sum())/3.0
	imuObj.Mags = ((maxvals - minvals)/2.0)*(1/averageRad)

	imuObj.setSRD(currentSRD)

print ("Mag calibration starting")
time.sleep(2)
caliberateMagApproxDebug(imu)
# imu.caliberateMagPrecise()
print ("Mag calibration Finished")
print (imu.MagBias)
print (imu.Magtransform)
print (imu.Mags)
```

Could check the output on this and see if there is a difference between max and min values captured during the calibration process? 

I believe it is a similar issue in caliberateMagPrecise(), it is getting a singular matrix, which is normally unlikely. 

Thank You
Niranjan

marcofariasmx, 2022-05-15:
Hello @niru-5 just tried your code but I am still getting the same error, could it have something to do with the registers numbers?? At the beginning I get the error "The name is wrong [117] The mag name is different and it is [0]"

Here is all the output:
[image]

I am using a GY-91 device which is supposed to have the MPU9250 with the AK8963 magnetometer. I read something interesting in this repo https://github.com/ricardozago/GY91-MPU9250-BMP280 that may have something to do with changing addresses but I still haven´t figured out where the problem is.

Thank you very much for your help.

marcofariasmx, 2022-05-15:
@niru-5 I also just tried with another GY-9250 module and got "The name is wrong [112]" but the same error at the end
[image]

marcofariasmx, 2022-05-16:
@niru-5 I think this might also be related to my issue since I only see the x68 address appear https://www.reddit.com/r/embedded/comments/kurlke/i2cdetect_not_detecting_ak8963_sensor_on_mpu9250/

niru-5, 2022-05-16:
Hi @marcofariasmx, 

Could you check if you are getting any mag values without calibration? This will show if the mag sensor is being accessed or not.

```

  
import os
import sys
import time
import smbus

from imusensor.MPU9250 import MPU9250

address = 0x68
bus = smbus.SMBus(1)
imu = MPU9250.MPU9250(bus, address)
imu.begin()
# imu.caliberateGyro()
# imu.caliberateAccelerometer()
# or load your own caliberation file
#imu.loadCalibDataFromFile("/home/pi/calib_real_bolder.json")

while True:
	imu.readSensor()
	imu.computeOrientation()

	#print ("Accel x: {0} ; Accel y : {1} ; Accel z : {2}".format(imu.AccelVals[0], imu.AccelVals[1], imu.AccelVals[2]))
	#print ("Gyro x: {0} ; Gyro y : {1} ; Gyro z : {2}".format(imu.GyroVals[0], imu.GyroVals[1], imu.GyroVals[2]))
	print ("Mag x: {0} ; Mag y : {1} ; Mag z : {2}".format(imu.MagVals[0], imu.MagVals[1], imu.MagVals[2]))
	print ("roll: {0} ; pitch : {1} ; yaw : {2}".format(imu.roll, imu.pitch, imu.yaw))
	time.sleep(0.1)
```

Thank You
Niranjan

locissigtech, 2022-05-25:
Hi. I think i'm having same issue.

I'm trying to read the MPU9250 values with a Raspberry Pi. When I run the command 'i2cdetect -y 1', it shows the 0x68 address of the accel/gyro, the 0x76 address of the BMP280 and other addresses of some ADS1115 that i have wired (0x48, 0x49, 0x4a and 0x4b). It does not show the 0x0C of the AK8963.

When i run the Basic Usage, at the imu.begin() it shows the same issues than the relationed with this post: name is wrong, mag is different, looks like it did not write properly....

I'm new to imu's sensor, and almost new to python, so i'm really lost. I've read about bypass through mode, Fuse ROM, but I'm not understanding anything.

Could someone help me please?
Thank you in advance!!!!

marcofariasmx, 2022-05-25:
@niru-5 Thank you so much for the help, I concluded it was a faulty sensor. Tried a new one and worked well.  @locissigtech maybe the same is happening to you.

locissigtech, 2022-05-25:
Thank you so much!!! I'm trying that tomorrow, hope that's the solution!!

korolaab, 2023-06-02:
@locissigtech Hi, have solve the problem?

locissigtech, 2023-06-02:
> @locissigtech Hi, have solve the problem?

No. I went back to MPU6050 and use a LIS3MDL for the magnetometer. Then I used Madgwick Orientation Filter for quaternions and euler angles. Thank you for asking!