niru-5/imusensor #20: Magnetometer Calibration Procedure Inquiry

chefseg, 2022-04-18:
Hello @niru-5,

I hope you are doing well!
I have a few questions regarding the Magnetometer calibration process - how exactly does one properly calibrate the Magnetometer? 
Your blog post states to rotate the sensor in a "figure 8" shape but how exactly is this done? Would you be able to demonstrate this using pictures or video for further clarity? 

After calibration of the Magnetometer, when the sensor is stationary and I run the Jupyter Notebook script, three separate ellipsoids appear in separate regions in the plot. Is that a result of a good calibration or does that show remaining undesired magnetic effects?

Thank you so much for your help!

marcofariasmx, 2022-05-15:
@chefseg where you able to fix this or achieve proper calibration? I'm still confused on how to do it properly

niru-5, 2022-05-15:
Hi @chefseg and @marcofariasmx 

I believe a bit more clarification is required for the figure 8 movement. 
The movement could be seen in the following videos 
[video 1](https://www.youtube.com/watch?v=sP3d00Hr14o&ab_channel=ecompass3) The user does it once, but in this situation a bit longer till all the values are received. 
[video 2](https://www.youtube.com/watch?v=-Uq7AmSAjt8&ab_channel=GadgetHacks) 0:45 Similar movement is shown here as well. 

I will try to create a better graphic and update the readme. 
During the writing of the repo, this was done by packaging the whole hardware into a small case. The power source needs to be a bit away from the mag sensor as it can have an effect on it. 

Coming to the question of three different ellipsoids, they are not calibrated. I have written a bit about the reasoning here -> [link](https://medium.com/@niru5/intro-to-inertial-measurement-unit-imu-part-1-47f19fc7d68d#:~:text=Hard%2DIron%20Interference,matrix(3*3).).

Thank You
Niranjan